lvgl/lv_esp_idf #289: Unable to build for ESP32 TTGO Module - lvgl_spi_driver_init

prash85, 2021-09-14:
Trying to build for ESP32 TTGO T module. In menuconfig, I am selecting preconfigured board "TTGO T-Display" however the build fails and throws error shown below. I tried  switching to a different board and the build was successful. Is there a fix for this?

**Hardware:**

TTGO T-Display ESP32

https://au.banggood.com/TTGO-T-Display-ESP32-CP2104-CH340K-CH9102F-WiFi-bluetooth-Module-1_14-Inch-LCD-Development-Board-LILYGO-for-Arduino-products-that-work-with-official-Arduino-boards-p-1522925.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=AU&utm_content=minha&utm_campaign=minha-au-en-pc&currency=AUD&cur_warehouse=HK&createTmp=1&ID=6310222&utm_source=googleshopping&utm_medium=cpc_bgcs&utm_content=tanya&utm_campaign=tanya-SSC-AUg-newcustom-ncv95-ca3-0908&ad_id=544188101514&gclid=Cj0KCQjwkIGKBhCxARIsAINMioJ75EOD7gjGLy_cTprU5meAHnIpDTJPNC2PfYcxlB09_NedrpnEuBUaAvETEALw_wcB

**ERROR Message :**

../components/lvgl_esp32_drivers/lvgl_helpers.c: In function 'lvgl_driver_init':
../components/lvgl_esp32_drivers/lvgl_helpers.c:111:26: error: 'TFT_SPI_HOST' undeclared (first use in this function); did you mean 'VSPI_HOST'?
     lvgl_spi_driver_init(TFT_SPI_HOST,
                          ^~~~~~~~~~~~
                          VSPI_HOST
../components/lvgl_esp32_drivers/lvgl_helpers.c:111:26: note: each undeclared identifier is reported only once for each function it appears in
[1220/1228] Building C object esp-idf/lvgl_esp32_drivers/CMakeFiles/__idf_lvgl_esp32_drivers.dir/lvgl_tft/st7789.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

spacewrench, 2021-12-27:
I hit this same error, but was able to fix it.  The problem is in the Kconfig definitions that get selected when you choose LV_PREDEFINED_DISPLAY_TTGO.  For one thing, the TFT_DISPLAY_OFFSETS don't get selected properly.  But the TFT_SPI_HOST error that you got was because the TFT SPI bus type (VSPI/HSPI/FSPI) selection was skipped if LV_PREDEFINED_DISPLAY_TTGO is selected.

I haven't figured out how to write the Kconfig properly so that LV_PREDEFINED_DISPLAY_TTGO automatically selects LV_TFT_DISPLAY_SPI_HSPI=y, but hopefully I can get it soon.

Anyway, here is a patch for ...components/lvgl_esp32_drivers/Kconfig:

[TTGO-LVGL-Kconfig-patch.txt](https://github.com/lvgl/lv_port_esp32/files/7781458/TTGO-LVGL-Kconfig-patch.txt)

spacewrench, 2021-12-27:
Damn, I don't know how to work git in this situation.  Perhaps this is a combined repository or something, with some parts from core LVGL and other parts from the ESP32 port?  I don't know how to alert the maintainers about this fix.

C47D, 2021-12-27:
Thanks for the patch @spacewrench, I'm away from my develop machine, but will take a look at your patch and try to add it to the Kconfig.

prash85, 2021-12-29:
> f

Thanks spacewrench. 
I checked the Kconfig file under:
`LV_PORT_ESP32/components/lvgl_esp32_drivers/Kconfig` 
and it contains following two lines:
`rsource "lvgl_tft/Kconfig"
rsource "lvgl_touch/Kconfig"`

How should I add your patch file to the project? Should I just copy the contents of your patch file and paste it in existing Kconfig ? When I do that the build process fails. 

Also, I noticed in menuconfig when I select a TTGO board, options to select VSPI or HSPI disappears so I am assuming its getting configured internally? 

Thanks

spacewrench, 2021-12-29:
The Kconfig patch is for the file under lvgl_tft (not lvgl_touch).  If you go into that directory:

cd LV_PORT_ESP32/components/lvgl_esp32_drivers/lvgl_tft

you should be able to patch as:

patch -p2 < TTGO-LVGL-Kconfig-patch.txt

The VSPI/HSPI selection is the problem, but I couldn't figure out how to make Menuconfig go from PREDEFINED_DISPLAY_TTGO to HSPI selection without a menu display.  With this patch, the VSPI/HSPI menu choice shows, but I think the default is correct (it should be HSPI).

C47D, 2021-12-29:
Can you try updating the `lvgl_esp32_drivers` component to its current master? I would do so, but I'm away from my development machine and will be able to do it until next monday.

spacewrench, 2021-12-29:
I'm lost with git & submodules & etc...I just cloned a fresh copy of lvgl/lv_port_esp32, but it has the same problem if you try to select the predefined configuration for PREDEFINED_DISPLAY_TTGO.
Hopefully @prash85 will be able to get his project going with my Kconfig patch.  I don't think there's any rush otherwise...enjoy your holiday!

prash85, 2021-12-30:
@spacewrench Perfect! the patch works. Thanks a lot.

closed 2021-12-30 by prash85

arktrin, 2022-01-04:
@spacewrench and @prash85 Please, confirm that there is no (or there is) any single pixel offset on the screen in all display orientations if you have TTGO T-Display boards.

spacewrench, 2022-01-04:
I changed the offset defaults when I was working on that section -- the numbers in LVGL were slightly different from what was in the Lilygo Arduino setup.  I'm 99% sure they're correct, but I'll make a test program tonight and send some pictures.

spacewrench, 2022-01-04:
Sorry, the offset I had changed to 52 should be 53 (maybe that was the original?)  Here is a photo of the test screen drawing I made with offset 53
[image]
.

arktrin, 2022-01-05:
@spacewrench, thanks a lot for feedback! [52/53px offset issue](https://github.com/lvgl/lvgl_esp32_drivers/pull/154) is real for 240x135px ST7789 based displays. I just want to confirm that specifically for TTGO T-Display boards cause they are predefined in lvgl esp32 drivers. You may try (if you want) portrait display orientation to see 53px offset.

spacewrench, 2022-01-06:
Whew, that's a really fiddly bit of configuration.  I modified my Kconfig to set the offset to 52 or 53 depending on whether the display is inverted.  I think these pictures show what you need to see
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[Kconfig.txt](https://github.com/lvgl/lv_port_esp32/files/7818605/Kconfig.txt)
.