lvgl/lv_esp_idf #290: lvgl / lv_port_esp32 port_v8 Compilation failed

geekheart, 2021-09-26:
When I tried to use the port_v8 branch, I found that the compilation failed. It seems that the submodule corresponds to the previous version.

C47D, 2021-09-27:
Yes, it is usable in v7, we're trying to get support for v8 "soon".

@kisvegabor 
We should make that annotation into the README I guess.

geekheart, 2021-09-27:
> Yes, it is usable in v7, we're trying to get support for v8 "soon".
> 
> @kisvegabor
> We should make that annotation into the README I guess.

Looking forward to your updates to v8

kisvegabor, 2021-09-27:
> We should make that annotation into the README I guess.

Yes, I agree.

DatanoiseTV, 2021-11-15:
Is there any update on this? I am working on a project that could largely benefit from v8 and porting it later on would be a bit of a pain.

DCSBL, 2021-11-24:
@DatanoiseTV Just looking into it, but it seems that Espressif got version 8.0.2 to work, see https://github.com/espressif/esp-box/tree/master/components/lvgl

tvanfossen, 2021-11-24:
I've running lvgl @ master with lv_esp32_port on an esp32 without any issues since July/August, the changelog for[ LVGL v8](https://github.com/lvgl/lvgl/blob/master/docs/CHANGELOG.md#v800-01062021) detailed all the necessary changes to get it running

lv_esp32_port I'm using is effectively the same as master, with some fixes for the st7789 driver (fork in an ado repo, not sure if those fixes were merged?)

Relevant init calls
[image]

sukesh-ak, 2022-01-31:
> @DatanoiseTV Just looking into it, but it seems that Espressif got version 8.0.2 to work, see https://github.com/espressif/esp-box/tree/master/components/lvgl

Espressif is using [ESP-LCD component](https://github.com/espressif/esp-idf/tree/master/components/esp_lcd) I believe for the drivers. And they don't have drivers for all displays. So that route does not make sense ignoring all the existing display drivers available.

Not sure how @tvanfossen got it working, since it does not even compile for me with the following error. 

```
In file included from ../components/lvgl_esp32_drivers/lvgl_helpers.c:10:
../components/lvgl_esp32_drivers/lvgl_helpers.c: In function 'lvgl_driver_init':
../components/lvgl_esp32_drivers/lvgl_helpers.h:57:25: error: 'LV_HOR_RES_MAX' undeclared (first use in this function); did you mean 'LV_HOR_RES'?
 #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
```

These definitions were removed from v7 kconfig and looks like not implemented as part of device selection defaults after that.

tvanfossen, 2022-01-31:
I should've given a bit more detail. The esp-box drivers are limited to the ST7789 display driver at the moment as @sukesh-ak  mentioned. Also of note in the esp-lcd driver is having parallel drivers available instead of just spi. Waiting on a new board before testing parallel improvements

You need to set CONFIG_CUSTOM_DISPLAY_BUFFER_SIZE in the sdkconfig to avoid the compile issues around LV_HOR_RES_MAX. LVGL ESP DRIVERS -> LVGL TFT Display Controller -> Custom Display Buffer

sukesh-ak, 2022-02-01:
> You need to set CONFIG_CUSTOM_DISPLAY_BUFFER_SIZE in the sdkconfig to avoid the compile issues around LV_HOR_RES_MAX. LVGL ESP DRIVERS -> LVGL TFT Display Controller -> Custom Display Buffer

This fixed that compiler error but found another one.
Looks like the [develop](https://github.com/lvgl/lvgl_esp32_drivers/tree/develop) branch has the fixes. Will test that and update here later.

sukesh-ak, 2022-02-01:
Somehow I feel everything is messed up with the v8 migration. 

I tried the following steps
1. Create a fresh ESP-IDF project
2. Add lvgl (from main repo) as an IDF component
3. Add lvgl_esp32_drivers (develop branch) as a IDF component =compiles fine

Now I need an example to test. Could not find one with the latest functions. Every one of the sample was throwing errors because some function names have also changed during the port and probably some issues with kconfig too.

@tvanfossen Do you have a working ESP-IDF+v8 sample which shows just a label?

When you enable touch, the I2C port settings won't show up, when you go back to menuconfig, there is a new option to enable and define the I2C pins.

I am using Wireless-Tag WT32-SC01 3.5" 480x320 board with ST7796s and FT6336U.

Tried v7 using the master and it works but extremely slow.
Tested the display using Arduino+TFT_eSPI and performs great.

sukesh-ak, 2022-04-16:
Finally found LovyanGFX the best drivers which works for both Arduino and ESP-IDF.

ESP-IDF & Arduino samples for WT32-SC01 using LVGL8 + LOVYANGFX 

Device : WT32-SC01 
Display / Touch Drivers : LovyanGFX
Widgets : LVGL 8.x

ESP-IDF => WT32-SC01 + LVGL8 + LOVYANGFX 
https://github.com/sukesh-ak/LVGL8-WT32-SC01-IDF

Arduino => WT32-SC01 + LVGL8 + LOVYANGFX 
https://github.com/sukesh-ak/LVGL8-WT32-SC01-Arduino

closed 2025-05-19 by stale[bot]