lvgl/lv_esp_idf #292: unknown option "rsource"

snailwft, 2021-11-02:
Hi, How can I solve this problem.

root@ubuntu:/home/share/esp32/esp_light/lvgl/lv_port_esp32# make menuconfig
Toolchain path: /home/snail/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
MENUCONFIG
/home/share/esp32/esp_light/lvgl/esp-idf/components/lvgl_esp32_drivers/Kconfig:4: syntax error
/home/share/esp32/esp_light/lvgl/esp-idf/components/lvgl_esp32_drivers/Kconfig:3: unknown option "rsource"
/home/share/esp32/esp_light/lvgl/esp-idf/components/lvgl_esp32_drivers/Kconfig:6: syntax error
/home/share/esp32/esp_light/lvgl/esp-idf/components/lvgl_esp32_drivers/Kconfig:5: unknown option "rsource"
/home/share/esp32/esp_light/lvgl/esp-idf/components/lvgl_esp32_drivers/Kconfig:13: syntax error
/home/share/esp32/esp_light/lvgl/esp-idf/components/lvgl_esp32_drivers/Kconfig:12: unknown option "rsource"

embeddedt, 2021-11-02:
It seems that you have the wrong version of kconfig-frontends on your machine. I think the `rsource` keyword is an ESP extension, but I could be wrong, so perhaps check that all of the ESP-IDF tools are in your PATH.

tore-espressif, 2021-11-16:
Could you please share your IDF version?
`rsource` was added in ESP-IDF 4.2

snailwft, 2021-12-31:
@embeddedt @tore-espressif Thanks, I update the IDF to solve.

closed 2021-12-31 by snailwft