lvgl/lv_esp_idf #306: Out of the box experience on ESP32-WROVER-KIT

arnoutdekimo, 2022-03-22:
The out of the box experience of following the README.md on the latest IDF is not ideal on the ESP32-WROVER-KIT. 

The build succeeds, but some config settings are wrong, causing a messed up image.

I had to manually change in the config:

In LGL config:

- Maximum horizontal resolution 320
- Maximum vertical resolution 240 (default was 480 causing interlace issue)
- Swap 2 bytes of RGB565 color (otherwise colors are messed up)

In TFT display controller:
- Select predefined display: ESP-WROVER-KIT
- Display orientation: landscape

Given that this kit is probably the "default" for testing out the ESP32 LCD, I would imagine this was a bit more straightforward (at least some reference in the readme)

(For the rest pretty nice port!)

nicklasb, 2022-04-13:
TBH, and no offense,  that's not much config at all, there are a *myriad* of possible hardware combos.
For example, the default is probably more common. 

I'd say you experienced the closest to an IPhone unboxing possible in the MCU-dev world. :-)

bobby-RAD, 2023-03-08:
Thank you @arnoutdekimo for the post.  I agree that it was misleading (to me at least) to have a pre-defined display that required additional settings to work properly.  You saved me a great deal of time.  Much appreciated!

closed 2025-05-19 by stale[bot]