lvgl/lv_esp_idf #307: V8 roadmap

nicklasb, 2022-04-13:
Hi!

I saw that there was a V7 milestone, how about creating a V8 milestone?

If a roadmap detailing what is left was made, perhaps more would be able to help out?

Thanks for a great library!

sukesh-ak, 2022-07-13:
You can use this for v8.x

Feel free to check and create conf file as per your hardware needs.

https://github.com/sukesh-ak/LVGL8x-SDSPI-Template

nicklasb, 2022-10-11:
> @nicklasb
> 
> You can use this for v8.x
> 
> Feel free to check and create conf file as per your hardware needs.
> 
> https://github.com/sukesh-ak/LVGL8x-SDSPI-Template

Oops, just saw this for some reason. 
Will take a look as soon as I continue that project.

closed 2025-05-19 by stale[bot]