lvgl/lv_esp_idf #311: White screen ST7789V

EtienneMdv, 2022-05-09:
Hi all,

I am trying to configure LVGL for my ESP32 custom board with a ST7789V TFT LCD driver. Unfortunately, I haven't been very successful so far... The backlight seems to work as the only thing I get is a white screen. I checked the GPIOs for the SPI numerous times and they are correct.

I don't know where to start to troubleshoot my issue. Do you have any idea?

I appreciate your help.

EtienneMdv, 2022-05-09:
By the way, I was successful using [ESP32_TFT_library](https://github.com/loboris/ESP32_TFT_library) prior to trying using LVGL.

tvanfossen, 2022-05-09:
If your backlight is turning on, then I'd recommend checking that the LVGL initialization is completing properly

EtienneMdv, 2022-05-09:
Thank you @tvanfossen.

Is there an efficient to check that? I have turned the logs to trace but I don't see anything alarming.

tvanfossen, 2022-05-09:
menuconfig -> LVGL config -> feature config -> logging -> enable log module
Set default log verbosity to Log Important Events
Log with printf

Is your code entering the main app_main loop? Can you see logs in a main loop? Are you running lv_task_handler consistently? How much memory are you providing for lvgls memory buffer?

EtienneMdv, 2022-05-09:
I set the log according to your guidance. My code is not entering the main app_main loop... (I am attempting to run the demo app). My code is actually stuck calling _lvgl_driver_init()_. I am providing the default size for LVGLs memory buffer, 12800 bytes. The last log I get on my terminal is related to changing the display orientation to _LANDSCAPE_.

EtienneMdv, 2022-05-09:
Sorry about that... I am using ESP-IDF v5 and a custom board with a ESP-WROOM-32D. I am also thinking about an error in my IO setup since I am running the unmodified code. I changed the SPI mode from 2 to 0 and now I have something displayed. It is not really working but it is a start. I attached a pic.
[image]

tvanfossen, 2022-05-10:
IDF v5 may not be supported with LVGL yet, I'd recommend rolling back to ~4.2 as recommended by the lv_port_esp32 readme

That said, probably not the issue here. It looks to me as if your color depth is set incorrectly for your display driver

closed 2025-05-19 by stale[bot]