lvgl/lv_esp_idf #314: ESP IDF v5 support

ammaree, 2022-06-15:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
Fail to rebuild app after IDF upgrade

**Code to reproduce the issue**
Nothing changed from previous working code other than IDF upgrade

**Expected Results**
Working, error free compile

**Actual Results**
Various error messages

**ESP32 Chip version**
ESP32 WROVER

**ESP-IDF version**
v5 (Master)

**Development kit used**
ESP-WROVER-DEVKIT v4.1

**Development machine OS**
MacOS

**Compilation warnings/errors (if available)**
`
/Users/andremaree/Dropbox/devs/ws/z-repo/lvgl_esp32/lvgl_esp32_drivers/lvgl_tft/esp_lcd_backlight.c:10:10: fatal error: driver/ledc.h: No such file or directory
   10 | #include "driver/ledc.h"

/Users/andremaree/Dropbox/devs/ws/z-repo/lvgl_esp32/lvgl_esp32_drivers/lvgl_tft/disp_spi.c:10:10: fatal error: driver/gpio.h: No such file or directory
   10 | #include "driver/gpio.h"

/Users/andremaree/Dropbox/devs/ws/z-repo/lvgl_esp32/lvgl_esp32_drivers/lvgl_tft/disp_spi.h:18:10: fatal error: driver/spi_master.h: No such file or directory
   18 | #include <driver/spi_master.h>
`

**If possible, copy the compilation log into a file and attach it here**

ammaree, 2022-06-21:
Any feedback appreciated. I have had a look and it appears that the PWM backlight support might need to be updated.

sukesh-ak, 2022-07-02:
@ammaree You should check Migration guide since there are some breaking changes

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/build-system.html

ammaree, 2022-07-04:
All the required changes have been made. If I remove the display/GUI component of the project it builds, links and runs reliably.

The changes required I believe, are in the lvgl component.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/build-system.html

sukesh-ak, 2022-07-05:
@ammaree 
The initial errors you reported says those folders are not found. So i am guessing the location of files have changed.

ammaree, 2022-11-29:
@kisvegabor
It seems that the ESP32 port for LVGL (this repo) is still fixed to a 2 year old version of LVGL. In order for us to make some decisions, and hopefully move forward, answers to a couple of questions will be highly appreciated.
1. What are the plans, if any, for updating this repo to support ESP IDF v5.x?
2. What are the plans, if any, for updating this repo to support LVGL v8.x?
3. If no plans for the above, any pointers to an example we can use to try and get the above working?

Thanks in anticipation...

sukesh-ak, 2022-11-30:
> 

ESP-IDF 5.x is going with a direction using the concept of BSP. 
You can see some of the implementations here.
https://github.com/espressif/esp-bsp

There is also a new driver architecture being discussed for lvgl next version [here](https://github.com/lvgl/lvgl/issues/2720)

kisvegabor, 2022-12-01:
Thank for tagging me.

I think the best would be to update this repo for LVGL v8 and ESP-IDF 4 and make LVGL v9 (under development) support ESP-IDF v5. LVGL v9 will include all the layers to connect LVGL and ESP-IDF by enabling a config in `lv_conf.h`.

What do you think about it?

sukesh-ak, 2022-12-02:
> Thank for tagging me.
> 
> I think the best would be to
> 
> 1. update this repo for LVGL v8 and ESP-IDF 4
> 2. make LVGL v9 (under development) support ESP-IDF v5.
> 
> LVGL v9 will include all the layers to connect LVGL and ESP-IDF by enabling a config in `lv_conf.h`.
> 
> What do you think about it?

LVGL 8.x is working fine with LovyanGFX driver, so I am fine for now. Hope Espressif populates all the display drivers in their bsp repo.

nicklasb, 2022-12-06:
Hi, having the same problem, not sure how to interpret the discussion, is there any solution to this issue?

sukesh-ak, 2022-12-07:
> Hi, having the same problem, not sure how to interpret the discussion, is there any solution to this issue?

What display and controller?

sukesh-ak, 2022-12-07:
> Hi, having the same problem, not sure how to interpret the discussion, is there any solution to this issue?

Here is the IDF 5.0 project working with LovyanGFX as the driver for LVGL 8.x

LovyanGFX driver details are in the config files under `main/devices` folder
https://github.com/sukesh-ak/ESP32-TUX

nicklasb, 2022-12-07:
> Here is the IDF 5.0 project working with LovyanGFX as the driver for LVGL 8.x
>
> LovyanGFX driver details are in the config files under `main/devices` folder https://github.com/sukesh-ak/ESP32-TUX

Hi, 
I use ILI9488 + XPT2046, so that should work, problem is I use PlatformIO which doesn't ship with the later ESP-IDF.
So I will try that out with plain ESP-IDF at some later point in time, but thanks anyway, didn't know about that driver.

Kind regards, Nicklas

sukesh-ak, 2022-12-07:
@nicklasb 
When ESP32-S3 came out, PlatformIO took a while to update. 
That's when I completely shifted to using ESP-IDF. 

PlatformIO had a benefit for library integration using the config file. But now with IDF 5.0 onwards, IDF is moving in that direction with https://components.espressif.com/

In the context of LVGL, IDF 5.0 is moving to have BSP (Board Support Package) for devices on the components website.
It's still new effort so only Espressif devices are available for now here
https://github.com/espressif/esp-bsp

sukesh-ak, 2022-12-07:
@nicklasb 
Your lcd + touch is already available as components.

ILI9488 => https://components.espressif.com/components/atanisoft/esp_lcd_ili9488
XPT2046 => https://components.espressif.com/components/atanisoft/esp_lcd_touch_xpt2046

nicklasb, 2022-12-07:
> @nicklasb When ESP32-S3 came out, PlatformIO took a while to update. That's when I completely shifted to using ESP-IDF.
>
> PlatformIO had a benefit for library integration using the config file. But [...]

It is also about the IDE integration and so on, but thanks for the info on the components!

nicklasb, 2022-12-07:
TBH, Espressif could just as well have supported PlattformIO instead of creating a proprietary and competing platform.

sukesh-ak, 2022-12-07:
> TBH, Espressif could just as well have supported PlattformIO instead of creating a proprietary and competing platform.

Not sure which one that is (`proprietary and competing platform`).
PlatformIO uses either ESP-IDF or ESP-ArduinoCore (built using ESP-IDF) behind the scene.
Espressif supports VS Code and Eclipse option.

I think we are hijacking this issue now :)

nicklasb, 2022-12-07:
> Check your platformio.ini file which shows framework = espidf

Ok, but again, the code I am writing is not only going to run on Espressif MCU:s as I carefully avoid vendor lock-in. 
So sometimes, it will say framework=somethingelse. 

PlatformIO provides me with a layer between me and the manufacturers, and that is very important to me, and obviously many other PlatformIO users.

sukesh-ak, 2022-12-07:
> Ok, but again, the code I am writing is not only going to run on Espressif MCU:s as I carefully avoid vendor lock-in. So sometimes, it will say framework=somethingelse.
>
> PlatformIO provides me with [...]

Guess you are using Arduino then which provides this layer using Wire and other standards protocols. This is, if the said microcontroller supports these standard protocols (in short Arduino support).

One benefit of PlatformIO => silently downloads framework/sdks when you install a platform in it. Definitely provides convenience. 

Feel free to open a discussion here, if you want to continue => https://github.com/sukesh-ak/ESP32-TUX/discussions
I don't want this issue to be hijacked :)

nicklasb, 2022-12-07:
Tech comes and goes, there are many exciting platforms besides Arduino and ESP32, and either way I would never write only for any singular tech. I am just saying the same thing again and again, basically, so let's end it here.

sukesh-ak, 2022-12-07:
> Tech comes and goes, there are many exciting platforms besides Arduino and ESP32, and either way I would never write only for any singular tech. I am just saying the same thing again and again, basically, so let's end it here.

Cool. In the PC world its possible but not that much in the MicroController world because hardware is `proprietary`.

Several of them support Wire protocol including Raspberry Pi (WiringPi) so some portion of code can be reused. Otherwise, if a controller runs FreeRTOS then a lot of things can be similar. For example, AzureRTOS (known as [ThreadX RTOS](https://sukesh.me/2020/06/18/microsoft-iot-strategy-and-platform-options/) earlier before MS purchased it) has a FreeRTOS layer which is useful for porting in the case of STM32 controllers, since they have moved to AzureRTOS. 

Espressif uses FreeRTOS and now there is an Amazon version (after Amazon took over - [FAQ](https://www.freertos.org/FAQ_Amazon.html)).

nicklasb, 2022-12-07:
If you wan't to end a conversation, you can't make broad comparisons with the "PC World" and tell me where things are at like that. :-)

I see no huge issues making MCU project pretty much hardware independent. 
The PC World is like it is that only because of the gigabytes of lines of compatibility layer code that is *everywhere*.  The hardware of that world with its huge amount of highly complex components is actually way more complex and diverse than the MCU world. Only web devs think that browsers float in mid air.

sukesh-ak, 2022-12-07:
> If you wan't to end a conversation, you can't make broad comparisons with the "PC World" and tell me where things are at like that. :-)
> 
> I see no huge issues making MCU project pretty much hardware independent. 

It already exist called Arduino using Wire protocol (like I mentioned earlier). Runs on several controllers already.

Hobby projects with single source files have been easy to make it work on different controllers.
If its otherwise, share a link of your project working on different micro-controllers so that I can educate myself.

nicklasb, 2022-12-07:
> Hobby projects with single source files have been easy to make it work on different controllers. If its otherwise, share a link of your project working on different micro-controllers so that I can educate myself.

Could we perhaps de-escalate a bit here? I am just stating my goals and ambitions, I am sorry if they offend you, but I am afraid I will keep them regardless. Also, please don't read any attitudes in what I am writing.

But I have been developing in anything from assembler to javascript for 35 years now, and code is code, and business is business. And I do know a bit of both. And because of that, perhaps I am ready to #if a bit more than most.

sukesh-ak, 2022-12-07:
Chill. You didn't offend me :)
Sounds like we started during similar timelines. I remember using inline assembly to move mouse cursor during MS-DOS/Win 3.x times. Also writing TSR programs in those days.

Good luck with your projects and I believe `you` can make it happen with enough #if like you mentioned.

ammaree, 2022-12-29:
> Thank for tagging me.
> 
> I think the best would be to
> 
> 1. update this repo for LVGL v8 and ESP-IDF 4
> 2. make LVGL v9 (under development) support ESP-IDF v5.
> 
> LVGL v9 will include all the layers to connect LVGL and ESP-IDF by enabling a config in `lv_conf.h`.
> 
> What do you think about it?

@kisvegabor 

I think that ESP-IDF V5.x support should be incorporated into LVGL v8.x ASAP.

The current lv_port_esp32 repo is already 2 years old and still tied to IDF V4 (and V3) as well as LVGL v7.9 (I think)

Considering the growing market share and acceptance of the ESP32 {S?/C?/H?] it will allow many developers to move to LVGL v8.x and IDF v5.x, possibly with optional v4.x support?

Personally, the sooner we can have IDF v5 with LVGL v8, the better

kisvegabor, 2023-01-02:
I definitely agree on updating to LVGL v8 and IDF v5, however unfortunately I have no capacity to work on it now. 

We have a [sponsorship program](https://github.com/lvgl/lvgl#heart-sponsor)  and I'd be happy to give a few hunderd USD if someone updates this (and maybe [lvgl_esp32_drivers](https://github.com/lvgl/lvgl_esp32_drivers too?) to LVGL v8 and IDF v5.  

Would you be interested in it?

zackees, 2023-01-06:
FYI: They upgraded v5 but didn't release an arduino port. I'm trying to port it now and it's a mess.

sukesh-ak, 2023-01-07:
> FYI: They upgraded v5 but didn't release an arduino port. I'm trying to port it now and it's a mess.

Arduino versions are generated automatically when ESP-IDF is released
https://github.com/espressif/arduino-esp32

zackees, 2023-01-07:
So the mainline supports v5.X and the releases only go up to 4.3 is that correct?

zackees, 2023-01-07:
The docs say it's only upto 4.3

ammaree, 2023-02-20:
@Maldus512

Apologies, should have been more clear, lv_port_esp32 is (was?) the problem. We have in the mean time changed to use LVGL 8.x (master) with LGFX (master) and that seems to work well. Ultimately the ideal would be latest LVGL with drivers offering:
> smallest memory (SRAM and flash) footprint
> highest refresh performance
> broadest MCU, display and touch controller support.

Most probably LGFX drivers with LVGL?

exislow, 2023-11-24:
Would be great, if somebody could port it to esp-idf v5

ammaree, 2023-12-20:
@kisvegabor

Another 10 months later, is there ANY working configuration of LVGL with ESP-IDF v5.1.x or later, preferably v5.2.x?

If not, any plans to ever support ESP-IDF version v5.x?

mhaberler, 2023-12-20:
@ammaree take this as a starting point: https://github.com/mhaberler/ESP32-TUX/commits/additional-platforms/

nicklasb, 2023-12-20:
> @ammaree take this as a starting point: https://github.com/mhaberler/ESP32-TUX/commits/additional-platforms/

@mhaberler I suppose that @ammaree asked not on how to create an Lvgl version for ESP-IDF 5 port himself. 
If not, my bad. However, that is my question, and if I understand @kisvegabor correctly, this is not in the works. 
(And nothing Squareline is considering either then, I suppose).

kisvegabor, 2023-12-20:
I'm not planning to update this repository, however if someone is interested in working on it, let me know.

We will release LVGL v9 on January 15, and from v9 we will move all display driver to the LVGL repo directly and I'm more interested in working on that.  If you would like to contribute to that, it would be more than welcome. If there are any special IDF v5  consideration, please let me know in an issue in the [main LVGL repo](https://github.com/lvgl/lvgl).

nicklasb, 2023-12-20:
> If there are any special IDF v5 consideration, please let me know in an issue in the [main LVGL repo](https://github.com/lvgl/lvgl).
I am not sure what you mean with IDF v5 considerations, are you saying that LVGL already supports ESP-IDF?
If so, I have missed that information completely. I can see that there is an ESP-IDF component now..but does that mean ESP-IDF and not something else?

If not, the problem I think is that people doesn't know where to start. 
Perhaps if you could help jump starting this in some way by creating an issue where you list what needs to be done?

kisvegabor, 2023-12-20:
Oh, maybe there is a fundamental misunderstanding here. So let's start from the beginning:
- LVGL v8.3 (v9 is not tested yet) works with ESP-IDF v5. I think using the ESP's component registry is a  good way to add LVGL to an IDF project.
- I don't use ESP32 on daily basis, so maybe I'm unaware of something regarding the requirement about the drivers. What I know is that Espressif has an LVGL layer which makes it simple to use Espressif's supported touch and display controller's with LVGL. See [here](https://github.com/espressif/esp-bsp/tree/master/components/esp_lvgl_port).
- In LVGL v9 we will have 
   - built in display drivers, see [this PR for ST7789](https://github.com/lvgl/lvgl/pull/5020) (unfortunately it will be merged only in January)
   - `esp_lcd` (the wrapper around ESP's abstract display interface) will be directly supported in LVGL as well. See [here](https://github.com/lvgl/lvgl/pull/4507) (also in January)
- Our docs about ESP is out dated: [v8](https://docs.lvgl.io/8.3/get-started/platforms/espressif.html), [v9](https://docs.lvgl.io/master/integration/chip/espressif.html)

nicklasb, 2023-12-20:
Wow, there certainly is a misunderstanding. And great news! 
That the PR won't come until January is hardly a problem, I have waited around here for a year now, have no issues with doing something else for a couple of more weeks. :-)

I think that some people end up here on the esp32 port and never leave, they aren't reached by any communication about mainline ESP-IDF support. And this thread might be a bit TLDR. I for example have chosen to work on other aspects of my stuff the last year, and this has been the only place I have seen updates. I guess I could have understand it from something above, however i don't seem to be alone in that.

Some proposals then:
1. On packages, please change the mention of "ESP32 component" into "ESP-IDF (ESP32) component" or something that indicates not only ESP32 but specifically ESP-IDF support. 
2. Also try say ESP-IDF, not ESP or IDF. As they won't help with internet searches.
3. On _this_ repository, make a huge banner on the front pages that redirects users to the man LVGL with some of the above information. 
4. Agreed on the docs, it would be great if at least the basic info that have come up here goes in there.

kisvegabor, 2023-12-21:
> On packages, please change the mention of "ESP32 component" into "ESP-IDF (ESP32) component" or something that indicates not only ESP32 but specifically ESP-IDF support.

Which "packages" do you mean exactly? Could you send a link?

> Also try say ESP-IDF, not ESP or IDF. As they won't help with internet searches.

Noted, thanks!

> On this repository, make a huge banner on the front pages that redirects users to the man LVGL with some of the above information.

I agree, once we will have those 2 PR merged, I will add the banner.

> Agreed on the docs, it would be great if at least the basic info that have come up here goes in there.

This is what I have in mind as a table of content. Please add anything that seems to be missing.
- General overview (1-2 paragraph) about what ESP32 and ESP-IDF is.
- Get started: Links the ESP docs to set up the environment
 - How to add LVGL? Component registry (or manually?)
 - Configure LVGL (menuconfig)
 - Basic LVGL integration: Tick interface, call `lv_timer_handler`
 - How to add display (and touch) drivers
    - Use LVGL's esp_lcd interface
    - Use esp_lcd directly
    - Use LVGL's display controller drivers
 - Example
    - How to connect a display with ILI9341 controller to an ESP32
    - How to add the write drivers     
    - Show a demo
 - Useful notes
    - How to optimize performance: e.g. which RAM for draw buffers, SPI clock, etc

nicklasb, 2023-12-21:
On the main README on the repo:
https://github.com/lvgl/lvgl
Under "Packages"

nicklasb, 2023-12-21:
Great!
> Which "packages" do you mean exactly?

On the main README on the repo:
https://github.com/lvgl/lvgl
Under "Packages" - "LVGL is available as:" - "ESP32 component". 
"ESP-IDF(ESP32) component" would have all the buzzwords and not mislead.

WRT headings, they look the part, just a small complaint would be about "Links the ESP docs", even though I hesitate,  as extrasensory perception sure would be useful in programming some times. :-)

kisvegabor, 2023-12-25:
Ahh, updated [here](https://github.com/lvgl/lvgl/commit/8e4b7f5676a2cb5bb52978974e237b0f746d76a1) :slightly_smiling_face: 

> WRT headings, they look the part, just a small complaint would be about "Links the ESP docs", even though I hesitate, as extrasensory perception sure would be useful in programming some times. :-)

Do you still mean in the README? If so, I think once we will have a better ESP-IDF docs on our website, we should link it from the README too.

nicklasb, 2023-12-25:
**LOOKING FOR ESP-IDF 5.x SUPPORT? DO NOT LOOK HERE BUT IN THE MAINLINE LVGL**

(It has been in there since v8.3, don't be like me and mope around in this repo forever)

@ammaree , @mhaberler