lvgl/lv_esp_idf #315: LVGL控件边缘出现莫名亮点

15172072715, 2022-06-25:
LVGL控件边缘出现莫名亮点，并且这样的问题会随着设置LVGL初始化时候的缓冲buff大小而改变点点的数量，当把缓冲减小的时候点点就会明显变多，buff越大，点点越小，我使用的是ESP32S3芯片，采用LVGL8.3模板，以下是buff设置大小不同的情况
[image]
[image]
[image]
[image]
[image]
[image]

15172072715, 2022-06-26:
[image]
无论是在guider上面还是用模拟器都不存在点点的情况

fred913, 2023-02-23:
As this is usually an issue tracker in English, it's absolutely suggested to speak English. Or more directly, you're supposed to speak English for better understanding from others, even using a translator.

asd3330303, 2023-08-09:
/************中文 start**********************/
参考(https://www.esp32.com/viewtopic.php?f=25&t=28324&p=118360#p118360)，已解决问题哈哈，这个问题卡了我半年
在函数void ili9488_flush(lv_disp_drv_t * drv, const lv_area_t * area, lv_color_t * color_map)
里最后
ili9488_send_color((void *) mybuf, size * 3);
ili9488_send_color((void *) (&mybuf[(size*3) - 3]), 3);//增加的，本质是将最后一个像素再发一次
heap_caps_free(mybuf);
/************中文 END**********************/

/************English start**********************/
thanks for(https://www.esp32.com/viewtopic.php?f=25&t=28324&p=118360#p118360)
The problem has been solved. The problem has stuck me for half a year
In function: void ili9488_flush(lv_disp_drv_t * drv, const lv_area_t * area, lv_color_t * color_map)
ili9488_send_color((void *) mybuf, size * 3);
ili9488_send_color((void *) (&mybuf[(size*3) - 3]), 3);//need add. Essentially, you send the last pixel again
heap_caps_free(mybuf);

/************English END**********************/