lvgl/lv_esp_idf #320: LVGL v8 status ? Animation with Square Line Studio

Olfox59, 2022-09-15:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
Hi All,

I'm starting a nexw project based on Lilygo T display ESP32S3 and LVGL.

I'm able to create a screen with the nice and usefull tool SquareLine Studio. I can display it on the devkit.

I tried to add an animation  on my image. I export the ui from sqaure line studio and tried to build on vscode. But call to function "" refer to undifined function.

After comparing v8 to v7 , i saw tat this function were not present in v7. So i need to go on v8 to benefit of it.

I tried to create a blank project, and add lvgl as a submodule to have v8 ready to use. But it seems that KConfig of lv_port_esp32 and last v8 lvgl are different. I don't find all the same options in menuconfig.

So should I only modify the Kconfig and add the missing options present in the lv_port_esp32 version may be? Or is there also other changes to do and it is more complex than that?

I saw the first ticket on this topics that exist for one year now but still open.

Thank you !

**Code to reproduce the issue**

**Expected Results**

**Actual Results**

**ESP32 Chip version**
ESP32S3
**ESP-IDF version**
v4.4.2
**Development kit used**
Lilygo T display ESP32S3
**Development machine OS**
Visual studio code
**Compilation warnings/errors (if available)**
implicit declaration of function 'lv_obj_get_x_aligned'; did you mean 'lv_obj_set_align'? [-Werror=implicit-function-declaration]
implicit declaration of function 'lv_obj_get_y_aligned'; did you mean 'lv_obj_set_align'? [-Werror=implicit-function-declaration]
implicit declaration of function 'lv_anim_set_user_data'; did you mean 'lv_obj_set_user_data'? [-Werror=implicit-function-declaration]
**If possible, copy the compilation log into a file and attach it here**

ammaree, 2022-11-11:
+1 for official LVGL v8.x support

MrMarteng, 2022-12-14:
+1 from me as well. 
Still trying to get LVGL running with lvgl and regular drivers library

kisvegabor, 2023-03-13:
I also got some customer request for ESP32-LVGLv8 support and I really like to have this and [lvgl_esp32_drivers](https://github.com/lvgl/lvgl_esp32_drivers) repos updated.

As v7 and v8 differ only in some minor API changes in the drivers I think it's not that difficult to update them. 

Unfortunately, I don't have enough hardware for a deep enough testing, however you might have already seen our [sponsorship program](https://github.com/lvgl/lvgl#heart-sponsor). From our donations I'd be happy to give **300 USD for updating these repos to v8**. Does it sound like a fair offer?

Would you be interested it? 

cc @C47D

ammaree, 2023-03-13:
With some help from @sukesh-ak I have changed to use LGFX Master with LVGL v8. Have a look at https://github.com/sukesh-ak/ESP32-TUX

So far that has worked very well with ESP-IDF v5.x but I have only tried it with ESP-WROVER-Kit v4.1 and the Makerfabs 16bit parallel+touch devkits.

The only changes I have had to make to get a clean compile are:
lv_demo_stress.c: lIne 77 change first %d to %lu
lv_example_table_2.c: line 95 change first %"LV_PRIu32" to %d

So the burning need from our side is gone and the $300 can be saved by using LGFX

C47D, 2023-03-14:
Hi @kisvegabor, thanks for CCing me but I'm pretty busy with day job in this days. Happy to see yall got it working!

closed 2025-05-19 by stale[bot]