lvgl/lv_esp_idf #329: failed to run idf.py menuconfig

Cun-Jiang, 2023-02-05:
**Describe the issue**
I follow the  README.md ,run git clone --recurse-submodules https://github.com/lvgl/lv_port_esp32.git.However,when i run idf.py menuconfig,something went wrong. And the folders `lvgl` and `lvgl_esp32_drivers` are empty。

`D:\ESPprojects\LVGL>cd lv_port_esp32

D:\ESPprojects\LVGL\lv_port_esp32>idf.py menuconfig
Executing action: menuconfig
Running cmake in directory d:\espprojects\lvgl\lv_port_esp32\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 d:\espprojects\lvgl\lv_port_esp32"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1")
-- Component directory D:/ESPprojects/LVGL/lv_port_esp32/components/lvgl does not contain a CMakeLists.txt file. No component will be added
-- Component directory D:/ESPprojects/LVGL/lv_port_esp32/components/lvgl_esp32_drivers does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at C:/Espressif/frameworks/esp-idf-v5.0/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'lvgl'.
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.0/tools/cmake/build.cmake:281 (__build_resolve_and_add_req)
  C:/Espressif/frameworks/esp-idf-v5.0/tools/cmake/build.cmake:574 (__build_expand_requirements)
  C:/Espressif/frameworks/esp-idf-v5.0/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:5 (project)

-- Configuring incomplete, errors occurred!
See also "D:/ESPprojects/LVGL/lv_port_esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1`
**ESP32 Chip version**
esp32s3
**ESP-IDF version**
esp-idf 5.0

MrSurly, 2023-03-20:
> And the folders `lvgl` and `lvgl_esp32_drivers` are empty。

This is the likely source of your error; those directories _should not_ be empty.  Are you _certain_ you did `git clone --recurse-submodules git@github.com:lvgl/lv_port_esp32.git`? 

In any case, try doing `git submodule update --init --recursive` in the `lv_port_esp32` top directory, and see if that fixes the issue.

NosakiTech, 2024-06-10:
> This is the likely source of your error; those directories _should not_ be empty. Are you _certain_ you did `git clone --recurse-submodules git@github.com:lvgl/lv_port_esp32.git`?
>
> In any case, try [...]

That worked for me! Thanks!