lvgl/lv_esp_idf #332: Display not showing the correct windows while running this example

TheElectronicMusicProject, 2023-04-24:
I'm using a 1.8TFT SPI 128x160 display.
I also use `develop/lvgl_7.11.0_idf_5.0` for lvgl_esp32_drivers and `v7.11.0` for lvgl.

**Describe the issue**
I use the following connections:

- PIN LED to 3.3V
- PIN SCK to GPIO14
- PIN SDA to GPIO13
- PIN A0 to GPIO2
- PIN RESET to GPIO4
- PIN CS to GPIO15
- PIN GND to ground
- PIN VCC to 5V

Running the example I don't see the correct result on the screen.

**Code to reproduce the issue**
The given example with the following [sdkconfig](https://github.com/lvgl/lv_port_esp32/files/11311872/sdkconfig.txt) where I matched the pins and the display resolution

**Expected Results**
The windows shown in the example

**Actual Results**
I see this picture on the screen with fluctuations of the content but it doesn't work.
[image]
In the serial monitor I don't notice anything "strange":
```
I (0) cpu_start: App cpu up.
I (297) cpu_start: Pro cpu start user code
I (297) cpu_start: cpu freq: 160000000
I (297) cpu_start: Application information:
I (301) cpu_start: Project name:     lvgl-demo
I (306) cpu_start: App version:      v2.0-68-gcffa173-dirty
I (313) cpu_start: Compile time:     Apr 24 2023 17:00:46
I (319) cpu_start: ELF file SHA256:  d5d3d50c2a493b36...
I (325) cpu_start: ESP-IDF:          -128-NOTFOUND
I (330) cpu_start: Min chip rev:     v0.0
I (335) cpu_start: Max chip rev:     v3.99
I (340) cpu_start: Chip rev:         v1.0
I (345) heap_init: Initializing. RAM available for dynamic allocation:
I (352) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (358) heap_init: At 3FFBAD68 len 00025298 (148 KiB): DRAM
I (364) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (370) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (377) heap_init: At 4008D2A0 len 00012D60 (75 KiB): IRAM
I (384) spi_flash: detected chip: generic
I (388) spi_flash: flash io: dio
I (393) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (40) lvgl_helpers: Display hor size: 128, ver size: 160
I (40) lvgl_helpers: Display buffer size: 5120
I (40) lvgl_helpers: Initializing SPI master for display
I (50) lvgl_helpers: Configuring SPI host HSPI_HOST (1)
I (60) lvgl_helpers: MISO pin: -1, MOSI pin: 13, SCLK pin: 14, IO2/WP pin: -1, IO3/HD pin: -1
I (60) lvgl_helpers: Max transfer size: 10240 (bytes)
I (70) lvgl_helpers: Initializing SPI bus...
I (80) disp_spi: Adding SPI device
I (80) disp_spi: Clock speed: 40000000Hz, mode: 0, CS pin: 15
I (90) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (100) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (320) ILI9341: Initialization.
I (540) ILI9341: Display orientation: PORTRAIT
I (540) ILI9341: 0x36 command value: 0x48
```

**ESP-IDF version**
I use ESP-IDF `v4.4.4-336-g4b404115e8`

**Development kit used**
ESP32-WROOM-32

**Development machine OS**
Windows 11

**Compilation warnings/errors (if available)**
Yes, some warnings at the bottom of the log.
[lvgl_log_build.txt](https://github.com/lvgl/lv_port_esp32/files/11311987/lvgl_log_build.txt)

**If possible, copy the compilation log into a file and attach it here**

1) Is this display supported?
2) If yes, what kind of error I'm encountering?
3) Is there a way to fix it?

Thank you for your help

CE3kmacriver, 2023-06-05:
Your Display looks like my does if I try to put the SPI frequency too high. Try backing it off.

TheElectronicMusicProject, 2023-06-12:
@CE3kmacriver I posted this issue [here](https://forum.lvgl.io/t/display-not-showing-the-correct-windows-while-running-this-example/11577/3), they said I should use another driver for this display. I looked for the ST7735S but it gives errors while compiling..