lvgl/lv_esp_idf #333: Sunton ESP32-8048S070 board

MMlodzinski, 2023-07-06:
Is there any chance for make a Sunton ESP32-8048S070 board working with this repository? It has ESP32-S3 chip, EK9716 display driver and GT911 touch driver.

Below link to its documentation and Arduino demo:
[Documentation and demo](http://pan.jczn1688.com/directlink/1/ESP32%20module/7.0inch_ESP32-8048S070.zip)

Aliexpress link to that board:
[Aliexpress](https://pl.aliexpress.com/item/1005004952726089.html?pdp_npi=3%40dis%21PLN%21115%2C02%20z%C5%82%21114%2C94%20z%C5%82%21%21%21%21%21%40211b5e2216885581205886728ea40d%2112000031135431788%21sh01%21PL%210&spm=a2g0o.store_pc_home.productList_2005205734546.subject_3)

mhaberler, 2023-07-06:
this example has a working demo for the 7" Sunton board:  
https://github.com/mhaberler/squareline-esp32/tree/main/src/sunton

I cobbled this together from various net finds

care for mutating this into an option for this repo?

MMlodzinski, 2023-07-06:
I saw those Arduino solutions before, but I wonder if that board could work with ESP-IDF?

mhaberler, 2023-07-07:
you can always build as "esp-idf with Arduino as component" - which should amount to the same effect

here's an example for a Core2 how to do this: https://github.com/mhaberler/espidf-arduino-blink/blob/main/platformio.ini - see the m5stack_bottom2 target

mhaberler, 2023-07-11:
care to post some example code?

MMlodzinski, 2023-07-11:
I used esp-idf 5.1. It has some examples, including a rgb_panel one. It is available in esp-idf/examples/peripherals/lcd/rgb_panel. In this example you have to modify rgb_lcd_example_main.c file located in main/rgb_lcd_example_main.c with following parts of code:

In lines 25-53:
```
#define EXAMPLE_LCD_PIXEL_CLOCK_HZ (16 * 1000 * 1000)
#define EXAMPLE_LCD_BK_LIGHT_ON_LEVEL 1
#define EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL !EXAMPLE_LCD_BK_LIGHT_ON_LEVEL
#define EXAMPLE_PIN_NUM_BK_LIGHT 2
#define EXAMPLE_PIN_NUM_HSYNC 39
#define EXAMPLE_PIN_NUM_VSYNC 40
#define EXAMPLE_PIN_NUM_DE 41
#define EXAMPLE_PIN_NUM_PCLK 42
#define EXAMPLE_PIN_NUM_DATA0 15  // B0
#define EXAMPLE_PIN_NUM_DATA1 7   // B1
#define EXAMPLE_PIN_NUM_DATA2 6   // B2
#define EXAMPLE_PIN_NUM_DATA3 5   // B3
#define EXAMPLE_PIN_NUM_DATA4 4   // B4
#define EXAMPLE_PIN_NUM_DATA5 9   // G0
#define EXAMPLE_PIN_NUM_DATA6 46  // G1
#define EXAMPLE_PIN_NUM_DATA7 3   // G2
#define EXAMPLE_PIN_NUM_DATA8 8   // G3
#define EXAMPLE_PIN_NUM_DATA9 16  // G4
#define EXAMPLE_PIN_NUM_DATA10 1  // G5
#define EXAMPLE_PIN_NUM_DATA11 14 // R0
#define EXAMPLE_PIN_NUM_DATA12 21 // R1
#define EXAMPLE_PIN_NUM_DATA13 47 // R2
#define EXAMPLE_PIN_NUM_DATA14 48 // R3
#define EXAMPLE_PIN_NUM_DATA15 45 // R4
#define EXAMPLE_PIN_NUM_DISP_EN -1

// The pixel number in horizontal and vertical
#define EXAMPLE_LCD_H_RES 800
#define EXAMPLE_LCD_V_RES 480
```

In lines 163-170:
```
// The following parameters should refer to LCD spec
            .hsync_back_porch = 16,
            .hsync_front_porch = 210,
            .hsync_pulse_width = 30,
            .vsync_back_porch = 10,
            .vsync_front_porch = 22,
            .vsync_pulse_width = 13,
            .flags.pclk_active_neg = true,
```

As soon as you do this, the example should start working.

mhaberler, 2023-07-11:
thanks - appreciated!

MMlodzinski, 2023-07-13:
I have written an example with this devboard and lvgl. For now only the LCD and touch are working, maybe in the future I will add some things like SD card reader or maybe I will write something ESP-BSP-like for that board, it depends on few things. Here is the link to my repo:
[Sunton ESP32-8048S070c LVGL example](https://github.com/MMlodzinski/Sunton_ESP32-8048S070c_ESP_IDF_LVGL_example)
I'll close this issue, as I've got what I wanted already. If someone has some questions about that repo in the link, please ask me there.

closed 2023-07-13 by MMlodzinski