lvgl/lv_esp_idf #335: Unable to compile on esp-idf version 5.0

tianrongqin, 2023-11-24:
When I build the item on esp-idf version 5.0,it can not build successfully.I know it can build on the esp-idf version 4.0,but I want to build on esp-idf version 5.0 successfully.Can you help me for this problem?

mohamedsiddik-b, 2023-12-27:
me also facing the same.

littleboot, 2024-01-09:
Same problem here, current stable version is IDF 5.1.2
Anyone found a fix yet?

littleboot, 2024-01-18:
Have a look at the pull requests someone has already fixed the issues so it can be compiled with IDF V5

LV port:
https://github.com/hiruna/lv_port_esp32/

Drivers
https://github.com/hiruna/lvgl_esp32_drivers/tree/develop/lvgl_8.3.7_idf_5.2

martinberlin, 2024-02-27:
@hiruna is not possible to make a Pull Request so this is also fixed here upstream?