lvgl/lv_esp_idf #338: Horizontal scrolling

liangfei2021, 2024-02-05:
1. TFT LCD(ST7789P3-240*284),
2. 8 pictures(190*190) are placed horizontally in the cont
3. then use lv_obj_set_scroll_snap_x(), lv_obj_scroll_to_view() to scroll horizontally, 
4. the FPS is about 10fps when scrolling, the effect is very bad!

BanMian11, 2024-02-26:
Hello, can you share the code that drives ST7789P3.

liangfei2021, 2024-03-13:
void st7789P3_init(void)
{
    lcd_init_cmd_t st7789P3_init_cmds[] = {
        {0xCF, {0x00, 0x83, 0X30}, 3},
        {0xED, {0x64, 0x03, 0X12, 0X81}, 4},
        {0xE8, {0x85, 0x01, 0x79}, 3},
        {0xCB, {0x39, 0x2C, 0x00, 0x34, 0x02}, 5},
        {0xF7, {0x20}, 1},
        {0xEA, {0x00, 0x00}, 2},
        {0xC0, {0x26}, 1},
        {0xC1, {0x11}, 1},
        {0xC5, {0x35, 0x3E}, 2},
        {0xC6, {0x0F}, 1},
        {0xC7, {0xBE}, 1},
        {0x36, {0x00}, 1}, // Set to 0x28 if your display is flipped
        {0x3A, {0x05}, 1},
        // {0x33, {0x00,0x46,0x00,0xD6,0x00,0x24}, 6},
        // {0x37, {0x00,0x45}, 2},
 
#if ST7789P3_INVERT_COLORS == 1
    {0x21, {0x00}, 0}, // set inverted mode
#else
    {0x20, {0x00}, 0}, // set non-inverted mode
#endif
 
        {0xB1, {0x44, 0x00,0x00, 0x14}, 3},
        // {0xB2, {0x0C, 0x0C, 0x00, 0x33, 0x33}, 5},
        // {0xB3, {0x11, 0x0F}, 2},
        {0xF2, {0x08}, 1},
        {0x26, {0x01}, 1},
        {0xE0, {0xD0, 0x00, 0x02, 0x07, 0x0A, 0x28, 0x32, 0x44, 0x42, 0x06, 0x0E, 0x12, 0x14, 0x17}, 14},
        {0xE1, {0xD0, 0x00, 0x02, 0x07, 0x0A, 0x28, 0x31, 0x54, 0x47, 0x0E, 0x1C, 0x17, 0x1B, 0x1E}, 14},
        {0x2A, {0x00, 0x00, 0x00, 0xEF}, 4},
        {0x2B, {0x00, 0x00, 0x01, 0x3f}, 4},
        {0x2C, {0x00}, 0},
        {0xB7, {0x07}, 1},
        {0xB6, {0x0A, 0x82, 0x27, 0x00}, 4},
        {0x11, {0x00}, 0x80},
        {0x29, {0x00}, 0x80},
        {0x00, {0x00}, 0xff},
    };
 
    //Initialize non-SPI GPIOs
    gpio_pad_select_gpio(ST7789P3_DC);
    gpio_set_direction(ST7789P3_DC, GPIO_MODE_OUTPUT);
 
#if !defined(CONFIG_LV_DISP_st7789P3_SOFT_RESET)
    gpio_pad_select_gpio(ST7789P3_RST);
    gpio_set_direction(ST7789P3_RST, GPIO_MODE_OUTPUT);
#endif
 
#if ST7789P3_ENABLE_BACKLIGHT_CONTROL
    gpio_pad_select_gpio(ST7789P3_BCKL);
    gpio_set_direction(ST7789P3_BCKL, GPIO_MODE_OUTPUT);
#endif
 
    //Reset the display
#if !defined(CONFIG_LV_DISP_st7789P3_SOFT_RESET)
    gpio_set_level(ST7789P3_RST, 0);
    vTaskDelay(100 / portTICK_RATE_MS);
    gpio_set_level(ST7789P3_RST, 1);
    vTaskDelay(100 / portTICK_RATE_MS);
#else
    st7789P3_send_cmd(st7789P3_SWRESET);
#endif
 
    printf("ST7789 initialization.\n");
 
    //Send all the commands
    uint16_t cmd = 0;
    while (st7789P3_init_cmds[cmd].databytes!=0xff) {
        st7789P3_send_cmd(st7789P3_init_cmds[cmd].cmd);
        st7789P3_send_data(st7789P3_init_cmds[cmd].data, st7789P3_init_cmds[cmd].databytes&0x1F);
        if (st7789P3_init_cmds[cmd].databytes & 0x80) {
                vTaskDelay(200 / portTICK_RATE_MS);
        }
        cmd++;
    }
 
    st7789P3_enable_backlight(true);
 
    st7789P3_set_orientation(CONFIG_LV_DISPLAY_ORIENTATION);
}

liangfei2021, 2024-03-13:
void st7789P3_flush(lv_disp_drv_t * drv, const lv_area_t * area, lv_color_t * color_map)
{
    uint8_t data[4] = {0};
 
    uint16_t offsetx1 = area->x1;
    uint16_t offsetx2 = area->x2;
    uint16_t offsety1 = area->y1;
    uint16_t offsety2 = area->y2;
 
#if (CONFIG_LV_TFT_DISPLAY_OFFSETS)
    offsetx1 += CONFIG_LV_TFT_DISPLAY_X_OFFSET;
    offsetx2 += CONFIG_LV_TFT_DISPLAY_X_OFFSET;
    offsety1 += CONFIG_LV_TFT_DISPLAY_Y_OFFSET;
    offsety2 += CONFIG_LV_TFT_DISPLAY_Y_OFFSET;
 
#elif (LV_HOR_RES_MAX == 240) && (LV_VER_RES_MAX == 240)
#if (CONFIG_LV_DISPLAY_ORIENTATION_PORTRAIT)
    offsetx1 += 80;
    offsetx2 += 80;
#elif (CONFIG_LV_DISPLAY_ORIENTATION_LANDSCAPE_INVERTED)
    offsety1 += 80;
    offsety2 += 80;
#endif
#endif
 
    /*Column addresses*/
    st7789P3_send_cmd(ST7789P3_CASET);
    data[0] = (offsetx1 >> 8) & 0xFF;
    data[1] = offsetx1 & 0xFF;
    data[2] = (offsetx2 >> 8) & 0xFF;
    data[3] = offsetx2 & 0xFF;
    st7789P3_send_data(data, 4);
 
    /*Page addresses*/
    st7789P3_send_cmd(ST7789P3_RASET);
    data[0] = (offsety1 >> 8) & 0xFF;
    data[1] = offsety1 & 0xFF;
    data[2] = (offsety2 >> 8) & 0xFF;
    data[3] = offsety2 & 0xFF;
    st7789P3_send_data(data, 4);
 
    /*Memory write*/
    st7789P3_send_cmd(ST7789P3_RAMWR);
 
    uint32_t size = lv_area_get_width(area) * lv_area_get_height(area);
 
    st7789P3_send_color((void*)color_map, size * 2);
 
}

liangfei2021, 2024-04-01:
@BanMian11