lvgl/lv_esp_idf #346: How do i reference this .h file in my main.c file?

keyone808, 2024-12-30:
**ESP-IDF -V5.3.2**
i want to use lvgl.h and st7789 driver in my project,but when i try to **git clone** it in my project, it did not work. I just try to git clone in my project, and #include "lvgl.h".

**here is my project dirctory tree:**
[image]
**here is my error:**
[image]
[image]

And there is others problem that my "menuconfig" have not LVGL TFT DISPLAY CONTROL. How could I do?

klaatu2a, 2025-03-04:
Its been 3 months and no response to your query.  There  is no action on this Github discussion.   I have run into this problem as well.  The whole path thing in VSC is a bit of a mystery to me.  I'm using the Micrsoft C/C++ Intellisense extension, which allows hyperlinking around within the code.  In this case, if you moused over this line in the file, it would show no path to the file.
I usually look at other includes and try to mimic them.  However, there are often multiple files with the same name, and including the right one is not obvious
Also, I have found that if a file becomes vistidual, from same a older version, but its pulled in with the make file, that can cause these errors too.    Try deleting the Build directory and empty the contents of the ui directory.

AdonaiDiazEsparza, 2025-03-10:
Hi, you are not adding the lv_port_esp32 as a component, you need to create a folder named "components" into the project, copy and paste there the component 
Then add the following line in the cmakefile inside of the main folder

list(APPEND EXTRA_COMPONENTS_DIR components)

it should work, but I was trying to use that components and they dont compile, I think is because are so old